owntracks/gw #96: offUntil must cause device to go silent

jpmens, 2015-01-04:
Setting `offUntil` in the future, currently is buggy: the device still *frequently* publishes `l`-type messages with the last time.

In addition, we have discussed that, when `offUntil` is set, it should also silence 

* trip
* cellinfo
* providerinfo
* GPIO
* obd2, FMS
* basically everything incl. start message

jpmens, 2015-01-04:
Alarms should still be reported probably

ckrey, 2015-01-04:
If offUntil is engaged, Greenwich will not transmit
- any location messages
- alarm message
- any voltage messages (might get info about start and stop engine from this)
- cellinfo message
- operatorlist message
- FMS or OBD2 messages

Still does send
- start message (version, IMEI and timestamp)
- $gps response with "no location available"
- debug messages which might contain private data
- output from commands via MQTT

Re Alarms:
What is the use case? Alarm is only activated when device is shutdown and motion is configured.

jpmens, 2015-01-04:
Sounds good. (Forget the alarms; you're right.)

ckrey, 2015-01-04:
If offUntil is engaged:
- lastFix is not stored/updated in settings when shutting down
- $state command response does not include CellInfo or Operators

ckrey, 2015-01-04:
If offUntil is engaged:

$state command response does Voltage info

ckrey, 2015-01-04:
If offUntil is engaged:

trip counter does not get updated

commit 67c70575 (2015-01-04): V0.11.42

offUntil revisited closes #96
closed 2015-01-04 by ckrey